PaulStoffregen/PS2Keyboard #20: ESP8266 and CORE_INT_EVERY_PIN

jscrane, 2018-09-04:
### Description

ESP8266 is only supported when ARDUINO_ESP8266_NODEMCU is defined. This isn't the case for, e.g., Wemos Mini D1.

### Steps To Reproduce Problem

Run Simple_Test example on Wemos Mini D1. It doesn't work. Change into_pins.h as follows and it works:

```
-#elif defined(ARDUINO_ESP8266_NODEMCU)
+#elif defined(ESP8266)
```

### Hardware & Software

Board Wemos Mini D1
Shields / modules used none
Arduino IDE version 1.8.4
Teensyduino version (if using Teensy) N/A
Version info & package name (from Tools > Boards > Board Manager) esp8266-arduino v2.4.2
Operating system & version Linux Mint Debian Edition
Any other software or hardware?

### Arduino Sketch

examples/Simple_Test.ino

### Errors or Incorrect Output

See above.

RickeyWard, 2019-01-06:
I can confirm, it does work on the wemos D1 mini with this change

PaulStoffregen, 2019-01-06:
Ok, done.

commit 34c7d047 (2019-01-06): Improve ESP8266 support, fixes #20
closed 2019-01-06 by PaulStoffregen