PaulStoffregen/PS2Keyboard #23: pin <-> interrupt mapping seems wrong on atmega1284

v0lker, 2018-12-13:
i connected CLK to the atmega1284 DIP's pin 3 (D2 / INT2), however the interrupt number calculated in PS2Keyboard::begin() was 0, not 2 as expected. took me an awfully long time to realise why `ps2interrupt()` wasn't called..

i got the desired value by using `digitalPinToInterrupt(irq_pin)` instead of what most of `begin()` does,  is there any reason not to use the arduino-provided function?

PaulStoffregen, 2018-12-13:
This library is very old, from long before Arduino had that function.

It should probably be updated.  I can tell you I'm not doing any non-critical updates to any libraries until at least mid-2019.  Currently working on a hardware project... the type that actually pays the bills, so I can then work on free software at other times.  Ping me on this in July 2019.

v0lker, 2018-12-13:
ah, ok, that makes sense. i'll send you a PR and you can merge it in july if you like...

v0lker, 2021-06-02:
how time flies - ping...