PaulStoffregen/PS2Keyboard #32: Ps2keyboard library not working on Arduino Zero

tinkerBOY-git, 2019-11-28:
Library doe not work with Arduino Zero. Is there a workaround?

no1so, 2020-01-26:
@tinkerBOY-git I am having the same problem trying to get this library working with a SAMD based board. Not a Arduino Zero but a Adafruit Feather M0 which uses the same board as the Zero. 

Did you ever find anything out regarding this?

tinkerBOY-git, 2020-04-18:
No, sorry.

MatthewBregg, 2021-06-15:
I'm using an M0, and got the library working.

The key was https://github.com/PaulStoffregen/PS2Keyboard/blob/13026d373f38da42ceda1de24ab15ea6405ef46c/PS2Keyboard.cpp#L707, as the M0 doesn't technically have all interrupt pins, (22 isn't), the earlier check https://github.com/PaulStoffregen/PS2Keyboard/blob/13026d373f38da42ceda1de24ab15ea6405ef46c/PS2Keyboard.cpp#L576 isn't defined, and the IRQ_NUM remains 255, so the interrupt is never attached.

For a quick hack, just remove the preprocessor directive at https://github.com/PaulStoffregen/PS2Keyboard/blob/13026d373f38da42ceda1de24ab15ea6405ef46c/PS2Keyboard.cpp#L576 and set irq_num = irq_pin unconditionally.

One suggestion: Could begin return a simple error code/bool indicating if the irq_num ever got set? Would make this a lot easier to debug and notice!

DavyVos, 2024-04-20:
@MatthewBregg thanks for your answer, I encountered the same problem using a stm32 nucleo 64 f303re board.
The issue indeed seems to be that irq_num is never set.