PaulStoffregen/PS2Keyboard #35: Wrong pinout on the help page?

marcin-chwedczuk, 2020-10-11:
According to picture that you put on: https://www.pjrc.com/teensy/td_libs_PS2Keyboard.html
The +5V line is on the right. This seems to be wrong, for example see:

[1] https://web.archive.org/web/20060901070855/http://www.computer-engineering.org/ps2protocol/
[2] https://en.wikipedia.org/wiki/PS/2_port

IMHO you should make it explicit if the pinout is for female or male connector.

PaulStoffregen, 2020-10-28:
Is this image from the web page incorrect?  Or is the connector's gender unclear?

[image]

DavidGari, 2024-01-18:
The pinout in your image is correct.  The connector is a Female 6-pin mini-DIN; the same one found on an IBM PS/2 computer, and many others.